fieldOfView/Cura-SidebarGUIPlugin #1: Messed positions

Liger0, 2019-04-01:
**Platform**
Windows 10 x64

**Actual Results**
The position of the words and graphics is messed up.

**Expected results**
Graphics should be kept in order.

**Additional Information**
[image]

fieldOfView, 2019-04-01:
Thanks for the report. For the top-most issue, could you check if that also happens if you set the language in Cura to english? If not, I'll know where to look.

Liger0, 2019-04-01:
Yeah, using the english Cura, the profile name tab appears correct.

fieldOfView, 2019-04-01:
Ok, I'll fix that in a future version. I will also make sure the extruder tabs use the same positioning logic as the extruder selectors in the original top bar.

closed 2019-06-11 by fieldOfView

fieldOfView, 2019-06-12:
Could you test this version:
http://files.fieldofview.com/cura/SidebarGUIPlugin-v6.0.0-2019-06-11T21_09_03Z.curapackage
Download the file, and drop it into a running Cura application window. Then restart Cura. It should work in both 4.0 and 4.1.

Liger0, 2019-06-12:
It doesn't seem to work on 4.1. I installed it and the only difference I see is that now the sidebar on the right has a different weird position, which is out of alignment on the left by default, tho I can move it.

fieldOfView, 2019-06-12:
There was a mistake in the original download. Can you download it again?

Liger0, 2019-06-12:
Yeah, it works nice! The extruder icons are also correctly centered vertically with a correct spacing while on stock cura they are partially hidden, so it's even better using this plugin.

It would just be a nice addition to horizontally center the extruder tabs.
[image]

fieldOfView, 2019-06-12:
> Yeah, it works nice!

👍 

> It would just be a nice addition to horizontally center the extruder tabs.

I don't agree.  Hardly anything is center-aligned in Cura, and I would like to keep it that way.

Liger0, 2019-06-12:
> I don't agree. Hardly anything is center-aligned in Cura, and I would like to keep it that way.

But the extruder tabs are centered like here :confused:
https://user-images.githubusercontent.com/20385116/59359243-4b215b00-8d2e-11e9-8075-85fe1af34f0b.png

Btw, I just wanted to make sure the names of the materials not being perfectly vertically-centered but a bit up was intended behaviour, just to avoid it to be emphatized (in a wrong manner) on different resolutions/Screen scale factors.

fieldOfView, 2019-06-12:
Yes, and that being the only center aligned things stands out like a sore thumb (IMHO). It will look even worse when the extruder icons are no longer in the center of the tab (because they are on the left of the material name). Finally, there's a technical challenge with having them center-aligned if you have - say - 4 extruders and the material names are long. Then you really want to have them left-aligned.