fieldOfView/Cura-SidebarGUIPlugin #2: Can't edit any fields after switching to Monitor Tab

SaltyLiquorice, 2019-05-08:
Just started today after I started a calibration print. Scaled my cube changed some settings in the sidebar then after slicing and hitting print went to the monitor tab to abort because I forgot to change temps.  Came back to the sidebar and I can't edit any of the input fields in the SidebarGui at all.

fieldOfView, 2019-05-10:
How do you connect your printer? USB, OctoPrint or Cura Connect?

SaltyLiquorice, 2019-05-10:
I have multiple printers some with octoprint, duet web control and an UM3.  The issue pops up with the cura connect monitor screen.  In fact today when I had a look the entire monitor tab wasn't even loading cura connect so maybe its an issue with cura itself.

OUPS65, 2019-05-12:
I have an Anycubic I3 mega and a MK3S. I connect both without problem via USB ( win family version 1809)

SaltyLiquorice, 2019-05-13:
Issue is gone now a restart of the PC fixed it.  Must have been a Cura problem.

camled, 2020-01-22:
I have the  same problem. With enabled Sidebar GUI plugin when after visit Monitor Tab I can't edit any text input in print settings. I can reproduce this behavior every time:
-Start Cura with enabled SidebarGUI plugin
-Go to Monitor tab
-Go back to Prepare or Preview tab
-Try edit any text input in print setting panel. It doesn't receive focus, cant type in it.

I tried reinstall plugin, reinstall Cura even reboot Windows ;)

fieldOfView, 2020-01-23:
Reopening the issue. @camled, what type of monitoring do you use? USB, Cura Connect or OctoPrint?

camled, 2020-01-23:
I use OctoPrint, but it hapens even when no online printer is used, just click on Monitor tab and back.
I just tried this on other computer:
-fresh Cura 4.4.1 install (with offline printer config: Creality Ender-3)
-no issue
-install SidebarGUI plugin (no other plugins)
-restart Cura
-click on Monitor Tab
-click on Prepare Tab
-text edit inputs stops receiving keyboard focus.

fieldOfView, 2020-01-23:
Thanks, I'll look in to it.

fieldOfView, 2020-02-05:
Could you try this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/SidebarGUIPlugin/WCLtDPLc7x6dkCsU4UI07vB52YrAitSeii1flDxCNyMr/SidebarGUIPlugin-v7.0.0-2020-02-05T22%3A05%3A10Z.curapackage?Expires=1896300314&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=dhkvlouhHsPYtSa16xaoUdepLOO9imCymLo6rCqS9s3aeB3Nx9mR4jupzGKeOruoX0FQUiZl8Vwn5krtJKR192TK5prP0Z5yhXUHQe0xR1%2FTBniYqMdSclo5jRayCb%2FlI3irz5zs%2BeMsGRps37oUpDHeDCkcUN4bTDnSsfdPWc7qRSNcQI%2FsUqAR6EwPIBtIFmzXIzzOBvUFSkpFakQ1KlVxAekIJ13XaGQuweve930cQp6J9SalxbyydA3emxpV2Qhd9bpWFAv3Qxv0ThBzHpiwy2AXN1p%2B9MayxD3VrS8JRWdKo8JShIPNt4rt2CUYLvdVSw6dXwf3%2Bp0fUXb7CQ%3D%3D)? Download the file, and drop it into the running Cura application window as if you are loading a model (not into a folder!). Then restart Cura.

camled, 2020-02-06:
This development version does not have this issue. I can edit text fields after switching from Monitor tab.

fieldOfView, 2020-02-06:
Great, thanks for testing. I have submitted a new version including this fix to the Marketplace.

closed 2020-02-06 by fieldOfView