fieldOfView/Cura-SidebarGUIPlugin #3: Cura 4.1 Mac not working

medoh300, 2019-05-30:
the plugin doesnt seem to work as before on 4.1 on mac , probably because of the resize windows feature of cura 4.1

fieldOfView, 2019-05-30:
The Sidebar GUI plugin is a plugin that will always need to be updated between versions. I do plan to update it for Cura 4.1, but I have not yet gotten round to it.

skyhisi, 2019-06-01:
I've found it's not working on Windows 10 either, the Cura log file reports:
```
DEBUG - [MainThread] SidebarGUIPlugin.__matchVersion [48]: This version of the plugin is not compatible with this version of Cura. Please check for an update.
WARNING - [MainThread] SidebarGUIPlugin.register [23]: Plugin not loaded because of a version mismatch
ERROR - [MainThread] UM.PluginRegistry.loadPlugin [392]: Plugin SidebarGUIPlugin did not return any objects to register
```

fieldOfView, 2019-06-01:
@skyhisi, you need to update the plugin. Please check the Marketplace.

fieldOfView, 2019-06-11:
A test-version is available here:
http://files.fieldofview.com/cura/SidebarGUIPlugin-v6.0.0-2019-06-11T21_09_03Z.curapackage
Download the file, and drop it into a running Cura application window. Then restart Cura.

01msmr, 2019-06-12:
4.0.1 also doesn't work on 10.14.5 (Mojave).

fieldOfView, 2019-06-12:
Sorry, I forgot to include a changed file. Could you try downloading the same file again? Thanks.
(This time I tested it myself...)

01msmr, 2019-06-12:
Thanks! Now it works.

closed 2019-06-12 by fieldOfView

fieldOfView, 2019-06-12:
The posted version is now published in the Marketplace.

01msmr, 2019-06-13:
Great! works as expected.