fieldOfView/Cura-SidebarGUIPlugin #4: [Feature Request] Floating sidebar !

yet-another-average-joe, 2019-06-02:
Hello, and thanks for this plugin.

(and waiting for 4.1 version)

The title could sound strange, as this plugin makes the sidebar not floating ! But maybe you coould add a feature that would allow the sidebar to be displayed out of the Cura window. This would be interresting for those who have multiple screens : Cura on one monitor, the sidebar on another.

fieldOfView, 2021-02-15:
I finally have some progress to report on this request:
[image]
There's still some work to be done, most notably fixing how tooltips get displayed; currently they are drawn in the main window instead of in the toolpanel.

yet-another-average-joe, 2021-03-13:
Can't believe it !
I've not been printing for 2 months. Today, I completed a design, ran Cura and first updated the plugins. At first I didn't notice the small buttons in the sidebar. And finally played with them : what are they ??? And discovered that the sidebar can now be undocked and moved to another monitor !
Many thanks ! This is the greatest Cura GUI enhancement ever. A life changer. It virtually doubles the viewport area when tweaking the options.

fieldOfView, 2021-03-13:
It took me a little while, but I got there in the end.

closed 2021-03-13 by fieldOfView