fieldOfView/Cura-SidebarGUIPlugin #7: SideBarGUI with cura-from-source (version="master") ?

agordon, 2019-08-30:
Hello,

I'm trying to get SidebarGUIPlugin to work with my Cura setup.
I'm using Debian 10 (buster).

When running official Cura 4.2 from AppImage, and with the plugin manually installed in `~/.local/share/cura/4.2/plugins/SidebarGUIPlugin` - it works great.

However, when I build Cura from source (with git version equivalent to 4.2),
and the plugin manually installed in `~/.local/share/cura/master/plugins/SidebarGUIPlugin` - the plugin is listed in the "marketplace -> installed" window, but is not working (or not activated?).

The only mentions of the plugin in the logs are:
```
2019-08-29 20:02:05,309 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [394]: Loaded plugin CuraEngineBackend
2019-08-29 20:02:05,318 - DEBUG - [MainThread] SidebarGUIPlugin.__matchVersion [29]: Running Cura from source, ignoring version of the plugin
```

Any ideas on how to make it work, or how to troubleshoot this issue ?

Thanks for an awesome plugin!
gordon

fieldOfView, 2019-08-30:
The sidebar plugin needs to load version-specific resources. When loading Cura from source, it always reports the version as "master", so the plugin has no way to know which version-specific resources it needs. That's why it says it is "ignoring the version of the plugin", but the line after that should say "Loaded plugin SidebarGUIPlugin" and the plugin should load just fine. I'll change the wording because I can see it is confusing. 

Long story short: the plugin should work with running from source with the 4.2 branch of Cura. The provided log excerpt does not indicate it doesn't. I need the rest of the logs.

agordon, 2019-08-30:
Thanks for looking into that.
A complete log is here: https://pastebin.com/bB36jwmy .

The exact cura version is `4.2.0-424-g845c2d41a` (from git).
Happy to try/test different things if you have suggestions.

fieldOfView, 2019-09-01:
Thanks for the logs. It does not show why the plugin is not working.
Have you tried installing the plugin through the Marketplace instead of manually?

r15ch13, 2019-09-01:
It doesn't work on my Linux machine also.
Added some logging output for the version numbers to `__matchVersion()`.
I could add more logger calls if you require more information. :smile:

OS: Manjaro
Cura: 4.2.1 (from offical Arch repo)

<details>
<summary>cura.log</summary>

```log
2019-09-01 21:13:30,537 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FileLogger
2019-09-01 21:13:30,551 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimpleView
2019-09-01 21:13:30,551 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin STLReader was already loaded
2019-09-01 21:13:30,551 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin OBJReader was already loaded
2019-09-01 21:13:30,552 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin STLWriter was already loaded
2019-09-01 21:13:30,552 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin OBJWriter was already loaded
2019-09-01 21:13:30,552 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin LocalFileOutputDevice was already loaded
2019-09-01 21:13:30,553 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin CameraTool was already loaded
2019-09-01 21:13:30,553 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin MirrorTool was already loaded
2019-09-01 21:13:30,553 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin TranslateTool was already loaded
2019-09-01 21:13:30,554 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin ScaleTool was already loaded
2019-09-01 21:13:30,554 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin RotateTool was already loaded
2019-09-01 21:13:30,555 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin SelectionTool was already loaded
2019-09-01 21:13:30,555 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin ConsoleLogger was already loaded
2019-09-01 21:13:30,555 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin UpdateChecker was already loaded
2019-09-01 21:13:30,556 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin LocalContainerProvider was already loaded
2019-09-01 21:13:30,556 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin FileLogger was already loaded
2019-09-01 21:13:30,556 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin SimpleView was already loaded
2019-09-01 21:13:30,608 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OctoPrintPlugin
2019-09-01 21:13:30,618 - WARNING - [MainThread] SidebarGUIPlugin.__matchVersion [47]: 4.0.0
2019-09-01 21:13:30,618 - WARNING - [MainThread] SidebarGUIPlugin.__matchVersion [48]: 4.2.0
2019-09-01 21:13:30,618 - WARNING - [MainThread] SidebarGUIPlugin.__matchVersion [49]: 4.2.1
2019-09-01 21:13:30,619 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SidebarGUIPlugin
2019-09-01 21:13:30,619 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin STLReader was already loaded
2019-09-01 21:13:30,620 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin OBJReader was already loaded
2019-09-01 21:13:30,620 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin STLWriter was already loaded
2019-09-01 21:13:30,620 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin OBJWriter was already loaded
2019-09-01 21:13:30,626 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeWriter
2019-09-01 21:13:30,636 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ImageReader
2019-09-01 21:13:30,643 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SupportEraser
2019-09-01 21:13:30,644 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin LocalFileOutputDevice was already loaded
2019-09-01 21:13:30,644 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin CameraTool was already loaded
2019-09-01 21:13:30,644 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin MirrorTool was already loaded
2019-09-01 21:13:30,645 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin TranslateTool was already loaded
2019-09-01 21:13:30,645 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin ScaleTool was already loaded
2019-09-01 21:13:30,645 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin RotateTool was already loaded
2019-09-01 21:13:30,645 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin SelectionTool was already loaded
2019-09-01 21:13:30,663 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimulationView
2019-09-01 21:13:30,668 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LegacyProfileReader
2019-09-01 21:13:30,673 - WARNING - [MainThread] UFPReader.<module> [10]: Could not import UFPReader; libCharon may be missing
2019-09-01 21:13:30,678 - WARNING - [MainThread] UFPReader.<module> [10]: Could not import UFPReader; libCharon may be missing
2019-09-01 21:13:30,678 - ERROR - [MainThread] UM.PluginRegistry.loadPlugin [392]: Plugin UFPReader did not return any objects to register
2019-09-01 21:13:30,691 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeReader
[... cut ...]
2019-09-01 21:13:40,273 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2019-09-01 21:13:40,274 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [61]: Updating QualityProfilesDropDownMenuModel.
2019-09-01 21:13:41,631 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [188]: Error check finished, result = False, time = 5.3s
2019-09-01 21:13:46,422 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
```
</details>

agordon, 2019-09-04:
> Have you tried installing the plugin through the Marketplace instead of manually?

Yes, same results - the original "floating" side bar is showing.

fieldOfView, 2019-09-13:
I have added some more logging to check where the initialisation of the plugin stops.

r15ch13, 2019-09-16:
New log:

<details>
<summary>Details</summary>

```
2019-09-16 18:00:48,829 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FileLogger
2019-09-16 18:00:48,835 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimpleView
2019-09-16 18:00:48,835 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin STLReader was already loaded
2019-09-16 18:00:48,836 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin OBJReader was already loaded
2019-09-16 18:00:48,836 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin STLWriter was already loaded
2019-09-16 18:00:48,836 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin OBJWriter was already loaded
2019-09-16 18:00:48,836 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin LocalFileOutputDevice was already loaded
2019-09-16 18:00:48,836 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin CameraTool was already loaded
2019-09-16 18:00:48,837 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin MirrorTool was already loaded
2019-09-16 18:00:48,837 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin TranslateTool was already loaded
2019-09-16 18:00:48,837 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin ScaleTool was already loaded
2019-09-16 18:00:48,837 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin RotateTool was already loaded
2019-09-16 18:00:48,837 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin SelectionTool was already loaded
2019-09-16 18:00:48,837 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin ConsoleLogger was already loaded
2019-09-16 18:00:48,838 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin UpdateChecker was already loaded
2019-09-16 18:00:48,838 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin LocalContainerProvider was already loaded
2019-09-16 18:00:48,838 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin FileLogger was already loaded
2019-09-16 18:00:48,838 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin SimpleView was already loaded
2019-09-16 18:00:48,902 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OctoPrintPlugin
2019-09-16 18:00:48,916 - DEBUG - [MainThread] SidebarGUIPlugin.SidebarGUIProxy.__init__ [13]: SidebarGUI proxy created
2019-09-16 18:00:48,918 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SidebarGUIPlugin
2019-09-16 18:00:48,918 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin STLReader was already loaded
2019-09-16 18:00:48,918 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin OBJReader was already loaded
2019-09-16 18:00:48,918 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin STLWriter was already loaded
2019-09-16 18:00:48,919 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin OBJWriter was already loaded
2019-09-16 18:00:48,926 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeWriter
2019-09-16 18:00:48,934 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ImageReader
2019-09-16 18:00:48,943 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SupportEraser
2019-09-16 18:00:48,943 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin LocalFileOutputDevice was already loaded
2019-09-16 18:00:48,943 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin CameraTool was already loaded
2019-09-16 18:00:48,943 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin MirrorTool was already loaded
2019-09-16 18:00:48,944 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin TranslateTool was already loaded
2019-09-16 18:00:48,944 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin ScaleTool was already loaded
2019-09-16 18:00:48,944 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin RotateTool was already loaded
2019-09-16 18:00:48,944 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin SelectionTool was already loaded
2019-09-16 18:00:48,965 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimulationView
2019-09-16 18:00:48,972 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LegacyProfileReader
2019-09-16 18:00:48,977 - WARNING - [MainThread] UFPReader.<module> [10]: Could not import UFPReader; libCharon may be missing
2019-09-16 18:00:48,981 - WARNING - [MainThread] UFPReader.<module> [10]: Could not import UFPReader; libCharon may be missing
2019-09-16 18:00:48,982 - ERROR - [MainThread] UM.PluginRegistry.loadPlugin [392]: Plugin UFPReader did not return any objects to register
2019-09-16 18:00:48,997 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeReader
2019-09-16 18:00:49,000 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Import error loading module AMFReader
2019-09-16 18:00:49,001 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2019-09-16 18:00:49,001 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/python3.7/site-packages/UM/PluginRegistry.py", line 539, in _findPlugin
[... cut ...]
2019-09-16 18:00:58,612 - DEBUG - [MainThread] cura.Machines.QualityManager.initialize [136]: Lookup tables updated.
2019-09-16 18:00:58,726 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2019-09-16 18:00:58,726 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [61]: Updating QualityProfilesDropDownMenuModel.
2019-09-16 18:01:00,477 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [188]: Error check finished, result = False, time = 5.7s
```

</details>

gergely-xyz, 2020-01-25:
I have the same problem on Manjaro with Cura 4.4.1 installed via pacman, but it works fine if I use the AppImage of the same version

fieldOfView, 2020-01-25:
If you have the same problem, I need the same information: logs. Please post a link to your cura.log.

gergely-xyz, 2020-01-25:
[cura.log](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/4112388/cura.log)
Here you go!

fieldOfView, 2020-01-25:
Thanks, I am looking in to it. What strikes me now is the order in which the plugins are loaded. It could be that because of that the plugin tries to replace the UI before the original UI is loaded, and when the original UI is loaded the sidebar UI gets replaced by the original UI (instead of the other way around).

On an unrelated note to both @gergely-xyz  and @r15ch13, you need to install the trimesh module (https://pypi.org/project/trimesh/). That should have been installed alongside Cura, so whoever made the package you are using has to fix that. This does not affect the Sidebar GUI though.

fieldOfView, 2020-01-25:
> What strikes me now is the order in which the plugins are loaded.

This seems to be the case for all three installations, so it is finally clear what is going on (even though I still don't have an installation that reproduces it). I'll get cracking on a workaround.

fieldOfView, 2020-01-25:
This should be fixed by https://github.com/fieldOfView/Cura-SidebarGUIPlugin/commit/f951d7787077487c4c26ecd53cbb98f9e90b1fe6

A quick fix is to replace `SidebarGUIPlugin.py` on your system with this version: https://raw.githubusercontent.com/fieldOfView/Cura-SidebarGUIPlugin/4.0/SidebarGUIPlugin.py. It should be in the plugins folder in your configuration folder.

Please let me know if it works.

closed 2020-01-25 by fieldOfView

agordon, 2020-01-26:
@fieldOfView Works for me! thanks so much!!

fieldOfView, 2020-01-26:
🍾 

Sorry it took me a while to figure this out, glad we got to the bottom of it.

gergely-xyz, 2020-02-20:
The update came trough with the plugin marketplace and everything works now. Thanks a lot! I much prefer the GUI with this layout!

commit f951d778 (2020-01-25): Fix loading sidebar gui on certain Linux installations

Delay registering replacement stages until after the original stages are created

Fixes #7