fieldOfView/Cura-SidebarGUIPlugin #10: Plugin not working with Cura 4.4

drzejkopf, 2019-11-19:
Cura 4.4 is crashing on startup with plugin installed, I'm getting following errors in Cura log:

`CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]: An uncaught error has occurred!
CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: Traceback (most recent call last):
CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "C:\Users\user\AppData\Roaming\cura\4.4\plugins\SidebarGUIPlugin\SidebarGUIPlugin\SidebarGUIProxy.py", line 26, in getExtruderHasQualityForMaterial
CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     material_manager = application.getMaterialManager()
CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: AttributeError: 'CuraApplication' object has no attribute 'getMaterialManager'`

It's related to (or I believe it is, I'm not a pro programmer, I just made a quick research)
https://github.com/RedSerenityDev/Cura/pull/496
https://github.com/Ultimaker/Cura/commit/9b836d95d9db1c2a08ed43a53af2487c140294ed

fieldOfView, 2019-11-19:
Thanks for the report. Strange that this did not get caught by both my testing and the Ultimaker Marketplace vetting. Must be dependent on the type of printer(s) configured in Cura.

drzejkopf, 2019-11-19:
@fieldOfView Ah, this is correct. I'm using custom printer definition which I keep in %appdata%/cura/.../definitions and I have also quality/material profiles for it, the problem was "has_machine_materials" line in printer definition metadata, after deleting it and keeping only "has_materials" line everything seems to be working well. And' it's weird because that line caused problem only with Sidebar GUI plugin, everything else was fine and was working properly. Sorry for bothering, I'm closing issue

fieldOfView, 2019-11-20:
It should still be fixed, but at least I now know it is not something *all* users will encounter.

fieldOfView, 2020-01-16:
Sorry, I lost track of this issue. It should be fixed in this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/SidebarGUIPlugin/4VYvwrzaCSFKGAQ9SxenMWq-T1ZpilCx1r8a2rOM0Zjg/SidebarGUIPlugin-v7.0.0-2020-01-16T15%3A45%3A56Z.curapackage?Expires=1894549562&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=SfOaznolGJB5b0OfCabtl9TEB7NipQbvYpjDquoFM%2B9K8NNKYZ1HZMB40DSNWAyDMfeIENxZypkVr6BbtVuYbh3Eo372EDbXBNTsLBju%2Bufl03Uvbj1ah5WIqBJxGFF0%2FtQ9uH42k4W8bYe9luzMIzTovCyzyrOxNQFxoVSY%2FUYtjeyz%2FTWNiNw9HCWi6J8fKrBkqjuLVgBY7enQmWV%2FHhapfusGreNxaP9dvOSt%2Fs5M4U9s7w%2FImhTBrvCJDK2eNAF597qsubD9Z%2BrHC1qPygAFhEKg9k3hvrDuLLZdD6o3KYdhLhZE0sepe3LiGxliFfAgExQ7CwV4lKa8HSsWsQ%3D%3D)

fieldOfView, 2020-01-16:
Sorry, please check *this* [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/SidebarGUIPlugin/4VYvwrzaCSFKGAQ9SxenMWq-T1ZpilCx1r8a2rOM0Zjg/SidebarGUIPlugin-v7.0.0-2020-01-16T15%3A55%3A17Z.curapackage?Expires=1894550220&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=lWVbl3VBntK4Ds38ZMKJspbEsEelKuE3hRGUCfQ4N3%2Bp00qd0xE9A7FB83lefk8W20JCzfSgAjDtav52pZELXbhNDtWzZzTgP890IJZVijYnU3TsTKhOxdcSaFqpzFZoBJmmSjV0kYCHetV4T%2FmvkxQqW0%2Bb8m%2F1nDTd7NeLXn37IFrofpij%2Bc4re47ptcTQJb0Cw4xWRMSpEnW9%2F8U6PEGURehNck5llspfK%2BfshS6u49EPlmxOwzTx%2BfAjHWEqakQgJ3ZmbKwFIxl1GGOVU%2B1ZtsczBwGRi1oPyFDBglsvJjixFNJZNEfo52plh90U2lOBIdpnX1xEN4pwceOwtw%3D%3D)

fieldOfView, 2020-01-23:
This should now be fixed with the version available in the Marketplace. If not, please let me know.

closed 2020-01-23 by fieldOfView