fieldOfView/Cura-SidebarGUIPlugin #11: AttributeError: 'CuraApplication' object has no attribute 'getMaterialManager'

tomac01, 2020-01-15:
Hi,

I use cura 4.4.1 and get the following error when the plugin is installed:

Traceback (most recent call last):
  File ".....\AppData\Roaming\cura\4.4\plugins\SidebarGUIPlugin\SidebarGUIPlugin\SidebarGUIProxy.py", line 26, in getExtruderHasQualityForMaterial
    material_manager = application.getMaterialManager()
AttributeError: 'CuraApplication' object has no attribute 'getMaterialManager'

Could you check the problem?

Thx

fieldOfView, 2020-01-16:
Duplicate of #10, but thanks for the reminder.

Could you check if this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/SidebarGUIPlugin/4VYvwrzaCSFKGAQ9SxenMWq-T1ZpilCx1r8a2rOM0Zjg/SidebarGUIPlugin-v7.0.0-2020-01-16T15%3A45%3A56Z.curapackage?Expires=1894549562&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=SfOaznolGJB5b0OfCabtl9TEB7NipQbvYpjDquoFM%2B9K8NNKYZ1HZMB40DSNWAyDMfeIENxZypkVr6BbtVuYbh3Eo372EDbXBNTsLBju%2Bufl03Uvbj1ah5WIqBJxGFF0%2FtQ9uH42k4W8bYe9luzMIzTovCyzyrOxNQFxoVSY%2FUYtjeyz%2FTWNiNw9HCWi6J8fKrBkqjuLVgBY7enQmWV%2FHhapfusGreNxaP9dvOSt%2Fs5M4U9s7w%2FImhTBrvCJDK2eNAF597qsubD9Z%2BrHC1qPygAFhEKg9k3hvrDuLLZdD6o3KYdhLhZE0sepe3LiGxliFfAgExQ7CwV4lKa8HSsWsQ%3D%3D) fixes the issue?

fieldOfView, 2020-01-16:
Sorry, please check *this* [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/SidebarGUIPlugin/4VYvwrzaCSFKGAQ9SxenMWq-T1ZpilCx1r8a2rOM0Zjg/SidebarGUIPlugin-v7.0.0-2020-01-16T15%3A55%3A17Z.curapackage?Expires=1894550220&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=lWVbl3VBntK4Ds38ZMKJspbEsEelKuE3hRGUCfQ4N3%2Bp00qd0xE9A7FB83lefk8W20JCzfSgAjDtav52pZELXbhNDtWzZzTgP890IJZVijYnU3TsTKhOxdcSaFqpzFZoBJmmSjV0kYCHetV4T%2FmvkxQqW0%2Bb8m%2F1nDTd7NeLXn37IFrofpij%2Bc4re47ptcTQJb0Cw4xWRMSpEnW9%2F8U6PEGURehNck5llspfK%2BfshS6u49EPlmxOwzTx%2BfAjHWEqakQgJ3ZmbKwFIxl1GGOVU%2B1ZtsczBwGRi1oPyFDBglsvJjixFNJZNEfo52plh90U2lOBIdpnX1xEN4pwceOwtw%3D%3D)

tomac01, 2020-01-16:
Excellent! Now it works perfectly! Big Thx!

fieldOfView, 2020-01-17:
Thanks for testing

closed 2020-01-17 by fieldOfView