fieldOfView/Cura-SidebarGUIPlugin #15: Settings Sidebar stays visible when Gcode file is opened

GregValiant, 2020-08-27:
This leaves all the settings available as well as the Save options.  I played with "Saving" the file and it made a real hash of things when it re-wrote the Gcode file.  That might have been do to the plugins I had loaded.  In the Zip file - 
[Saved Gcode Files.zip](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/5136585/Saved.Gcode.Files.zip)
"PreSave.Gcode" is the file I opened.  "PostSave.gcode" is the result after saving with Cura.

fieldOfView, 2020-08-27:
>  it made a real hash of things

Could you elaborate? What went wrong?

GregValiant, 2020-08-28:
That was my technical explanation of what I saw in my quicky initial comparison of the two gcode files (contained in the zip file).  I just took another look.
I use the plugin "Show Progress" and it was active when the PreSave file was created.  In the PostSave file, a new set of M117 lines was added and they are all in Layer:0.
The PreSave file contains 219930 lines.  The PostSave file contains 220349 lines.  The difference of 419 is right there with the number of layers (417).  Those two items lead me to believe it was the Show Progress plugin that caused the problem.
When reading the files into AutoCad, the wireframes look identical so it appears that the Gcode itself remains unchanged.
It's likely that my initial response was over-blown.  Please delete the term "made a real hash of" and replace it with "changed".
It does appear that the plug-in was the issue.
I think that if a Gcode file is opened in Cura that it should be read-only.

fieldOfView, 2020-08-28:
Thanks for elaborating.

> I use the plugin "Show Progress" 

Is this a plugin or a post processing script? Do you have a link for me?

Normally for postprocessing scripts, there is comment added in the gcode that is checked before the postprocessing is applied again, for this exact reason. Once a postprocessing script is run on the gcode, it will not be run again on the same gcode.

> I do think that if a Gcode file is opened in Cura that it should be read-only.

I agree that the sidebar should not let the user edit the settings, because Cura cannot reslice from the gcode alone. However it should be possible to print the gcode, and in Cura printing and saving are "the same thing" so I can't really disallow saving and not disallow printing.

GregValiant, 2020-08-28:
https://github.com/louiswouters/3D-Printing/search?q=show+progress&unscoped_q=show+progress

Originally I had requested an option that Cura start numbering a gcode file from layer 1.  I do a lot of "pause at height" to make color changes and it is always a bit clumsy trying to remember if I should add a number or subtract a number when going from Gcode to Cura (or vice versa).  That postprocessor had issues when loading a Gcode file that started at 1 and so it was abandoned.
A second request was that the LCD display the layer number.  It's easy enough to figure out from the Z height, unless the layer height is an odd number like .17. Louis wrote a second postprocessing script that shows the Cura layer numbers on the LCD as well as the time remaining for the print.   He called that one "Show Progress".  I don't know if it's an official script.  I use it because now when I get out of my chair, my calculator doesn't fall out of my lap.

fieldOfView, 2020-08-28:
Ok, it is a postprocessing script. It should not be able to run twice on the same gcode; I'll see what is going on.

fieldOfView, 2020-10-30:
The part of this report that can be fixed with this plugin is fixed and the fix is available from the Marketplace. So this issue can be closed.

commit 33ba6ea7 (2020-10-01): Hide/disable print setup when loading sliced data

Fixes #15
commit f03f3d6f (2020-10-01): Hide/disable print setup when loading sliced data

Fixes #15
closed 2020-10-30 by fieldOfView