fieldOfView/Cura-SidebarGUIPlugin #17: Not compatible with Cura 4.8

5axes, 2020-10-26:
Release of Cura limited in the plugin.json file 
    "minimum_cura_version": "4.0",
    "maximum_cura_version": "4.7",
Is it normal ?

fieldOfView, 2020-10-26:
Yes, this is normal. The plugin does a lot of patching of qml files, and these patches may break Cura if the original qml files change. This is why each plugin release only works with the versions of Cura it is tested with.

The master branch has been made compatible with Cura 4.8, and is currently available as a beta version here:
https://community.ultimaker.com/topic/26014-sidebar-gui-for-cura-4/?do=findComment&comment=275194

I'll leave this issue open until the 4.8-compatible version is available in the Marketplace.

Thisismydigitalself, 2020-10-30:
Same here. I understand this is fixed for the released version.
https://github.com/Ultimaker/Cura/issues/8664

fieldOfView, 2020-10-30:
Did you read my comment just above?

Thisismydigitalself, 2020-10-30:
Hi, I am not sure i understand what are my options now.
I do understand this is fixed now but not sure if i can test it with 4.8b or i need the master version. and, can i install plugs which are not yet listed in the marketplace without login in?

fieldOfView, 2020-10-30:
> The master branch has been made compatible with Cura 4.8, **and is currently available as a beta version here**:
> https://community.ultimaker.com/topic/26014-sidebar-gui-for-cura-4/?do=findComment&comment=275194

fieldOfView, 2020-10-30:
The 4.8-compatible version is now available from the Marketplace.

closed 2020-10-30 by fieldOfView

Thisismydigitalself, 2020-10-30:
Perfect. 
Can't see myself using Cura without some of your must have plugins. thank you and keep up the good work.

[image]