fieldOfView/Cura-SidebarGUIPlugin #18: Feature requests for SideBarGuiPlugin

Thisismydigitalself, 2020-10-30:
I wonder if you can add the following features which i have asked the Cura team maybe 3 years ago to no avail and that is to add click buttons on this side bar which when clicked - all sub-settings categories pull-downs collapses and/or expended.
Second toggle will show only my most used settings - like a favorite list of settings within the different categories. 
Each time i slice a complex model which takes many hours to print i spend a lost of time expanding and collapsing so many settings and just before printing i, again, open all categories just to make sure all are set as needed. 
I have asked many years ago to be able to expand/collapse so going over all settings will be faster and less tedious, and having a favorite category which shows only my most important setting (meaning there will be a need to add a checkbox for each setting within my global custom selections) so it will be like a summery confirmation before sending the gcode to printer.

Forgive my poor Illustration.
[image]
[image]

Cheers, and keep up your amazing contribution to Cura. much appreciated.

fieldOfView, 2020-10-30:
I've been considering doing something like that, but it will probably be a different plugin so it can be used also by people who prefer the floating settings pane (apparently they exist). Might be named "Settings Accordion".

Thisismydigitalself, 2020-10-31:
Great, I have trust in you. I as well not sure who uses 4.x built in side bar but, anyways..
The important thing is, if possible, to have these two features bundled together for maximum effectiveness.
It will be great to have the expand/collapse all but together with adding this another group/category/tab which will show my most crucial settings at a glance - that would be super cool. it goes without saying that if Cura would have used tabs instead of pull down menus for each category there was no need to have this plugin cause each tab closes all others, but tabs has their cons as well. the main objective I am aiming for  is speed of use and clear confirmation for the user to know, just before printing that crucial settings are in order and nothing was clicked/checked/selected by accident (sometimes selected settings changes by accident while your mouse is moved away).

This kind of reminds me this thread of mine from few years back when I have asked the guys at Ultimaker to consider adding a feature to Cura where the user can run multiple configurable slicing profiles at once / sequentially which will then be presented to the user to evaluate which one suits the job best - such profiles could be for best strength, best looks, lightest print, infill angle, and so on. back than i was told this would have a huge performance load on the system but i think this can be cool.

Anyways,
Just my might-be-out-of-subject two cents.
Cheers.

yet-another-average-joe, 2021-04-23:
A similar a feature was requested about Cura itself : see https://github.com/Ultimaker/Cura/issues/7847

(I don't know what a backlog is !)

fieldOfView, 2021-04-26:
The "backlog" is the long list of things that developers hope to get round to one day.

fieldOfView, 2023-07-07:
I'll close this issue in lieu of https://github.com/fieldOfView/Cura-TabbedSettingsPlugin/
It is not quite the same, but fulfills a similar role.

closed 2023-07-07 by fieldOfView