fieldOfView/Cura-SidebarGUIPlugin #21: Compatibility with RawMouse plugin

smartavionics, 2021-04-22:
A user of the RawMouse plugin reported an issue (https://github.com/smartavionics/RawMouse/issues/14) which appears to be due to the presence of the SidebarGUI plugin.

While the 3d mouse is being manipulated, RawMouse may use the setActiveView() API function to switch the view from the normal simulated view to a quicker to render "fastview" thus improving the frame rate and making the display smoother. When the mouse becomes inactive, the view is switched back to the simulated view. This functionality appears to break when the SidebarGUI plugin is active.

This issue has been created to explore what is going wrong and provide a place to discuss possible changes to either or both of the plugins.

To give you an idea of what is being done, here's a snippet of code from RawMouse that does the view change while processing mouse movement events...
```
                if self._auto_fast_view or ctrl_is_active:
                    if self._controller.getActiveStage().getPluginId() == "PreviewStage" and self._controller.getActiveView().getPluginId() != "FastView":
                        self._controller.setActiveView("FastView")
                        self._fast_view = True
                elif self._fast_view:
                    self._controller.setActiveView("SimulationView")
                    self._fast_view = False
```

fieldOfView, 2021-04-23:
Thanks for the issue. From a cursory look it should be fairly simple to support this. I'll look in to it next week.

smartavionics, 2021-04-23:
Thanks.

RawMouse also has the ability to toggle between the preview and prepare stages using a spacemouse button and the code snippet that does that is:

```
            elif self._button_work["toggleview"]:
                if self._controller.getActiveView().getPluginId() == "SimulationView":
                    self._controller.setActiveStage("PrepareStage")
                    self._controller.setActiveView("SolidView")
                else:
                    self._controller.setActiveStage("PreviewStage")
                    self._controller.setActiveView("SimulationView")
```

fieldOfView, 2021-04-26:
This should be fixed with the above commit. It would be appreciated if you could give it a go.

smartavionics, 2021-04-27:
Hi Aldo, I've just tried out your master branch and the interaction with RawMouse appears OK to me. The toggling between the prepare and preview stages and the temporary flip to FastView when moving/zooming the mouse works as expected. So, looking good. Thanks for the swift solution.

discip, 2021-05-19:
@fieldOfView 
Good afternoon Sir,
maybe I it, but when are you going to update your plugin?
thanks in advance
kind regards

fieldOfView, 2021-06-11:
I have submitted a version that includes the fix mentioned above to the Marketplace. It is awaiting approval now.

discip, 2021-06-11:
@fieldOfView
## Thank you Sir! 😃

smartavionics, 2021-06-11:
Happy days, let's close this now, thanks.

fieldOfView, 2021-06-12:
It is my habit to keep the issue open until the fixed version is available in the Marketplace.

commit 224416f6 (2021-04-26): Fix interop with FastView (part of the RawMouse package)

Fixes #21
closed 2021-06-28 by fieldOfView