fieldOfView/Cura-SidebarGUIPlugin #24: Cura Slower/freezing when Cura-Sidebar is activate

Deed01, 2021-04-30:
Using Cura 4.9

Working like a charm if Cura-Sidebar is deactivate
If Cura-Sidebar activate then it become really slow, have to wait 10-30s after  many action ( getting the waiting mouse icon and/or "not responding" text in windows title) when for exemple:
- getting focus on cura (switching back from another software)
- click on a toolbar menu (to open menu)
- click on an item in toolbar menu (to get setting window for exemple)
- opening market place
- etc. 

Attached to this bug report:

Cura log when freezing (extension activate)
[Cura 4.9 Log with extension Cure-SideBAr_KO.log](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/6403174/Cura.4.9.Log.with.extension.Cure-SideBAr_KO.log)

Cura log + conf filles without plugin (work well)
[Cura 4.9 Log without extension Cure-SideBAr_OK.zip](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/6403175/Cura.4.9.Log.without.extension.Cure-SideBAr_OK.zip)

Sequence of Action in both record:
- Launch Cura
- Load an STL
- Move the 3D view
- click on toollbar
- oepn parameter window then close it
- open market place then close it
- Mve the 3D view

ps: got OctoPrint Connection extension 3.5.18 and UM3 Network Printing 1.0.1 activate: got WAN access throught WIFI and mounted drive to a NAS (did saw cura crawling them using FileMon from SysInternal to monitot CURA access)(no difference with or without x: drive => used a subst to test this)

ps: done in link to this issue https://github.com/Ultimaker/Cura/issues/1602

fieldOfView, 2021-04-30:
Thanks for the extensive issue. I'll try to find out what is going on. 

If I have a test-version, would you be willing to see if it fixes things?

fieldOfView, 2021-04-30:
The first log you posted shows only 2 seconds worth of logs. It's the same thing repeated over and over for 700 times, but there is no context so I can't do much with it. Could you try to create a new log?

fieldOfView, 2021-04-30:
It would be appreciated if you could help me test a fix.

Close Cura and replace `C:\Program Files\Ultimaker Cura 4.9.0\resources\themes\cura-light\styles.qml` with this version: https://raw.githubusercontent.com/Ultimaker/Cura/04bfd6b820cb5c3f5583a4830b5b4e0ccde27a46/resources/themes/cura-light/styles.qml. Then restart Cura, reenable the Sidebar GUI plugin and restart Cura once more.

Deed01, 2021-04-30:
Didn't correct it

Here a new log
[cura.log](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/6405125/cura.log)

And i try a small video to show you
https://user-images.githubusercontent.com/81071090/116692739-1f103100-a9bd-11eb-87d6-487348a0c06c.mp4

00:05 click on preference menu => appear at 00:21
00:13 starting to trying to move the view => starting to reactr at 00:29 then its fluent  (got a GTX 1080)
00:48 clicking again on different menu in toolbar with no reaction
01:35 asking to get Preference panel => finally got it working after the end of the video (took more than 2 minutes => then i can navigate inside fluently

Deed01, 2021-04-30:
And a small video without the plugin (it s fluid on my screen)(video is only 10fps)

https://user-images.githubusercontent.com/81071090/116694156-36e8b480-a9bf-11eb-8533-f239123ca8b0.mp4

fieldOfView, 2021-04-30:
Is the new log you provided with of without the patched `styles.qml`?

Deed01, 2021-04-30:
with it

Deed01, 2021-05-01:
Made  a record with a better tool (screento GIF)
add a tail on the log files to get info synchronised to user action (and timing)

[image]

Corresponding log file
[cura.log](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/6409551/cura.log)

And a record Without Sidebar GUI => everything works well
[image]

fieldOfView, 2021-05-01:
Thanks for the extensive reporting. I still can't reproduce the issue, but will keep looking for solutions by cleaning up code.

fieldOfView, 2021-05-01:
On a tangent, for me opening the Marketplace and waiting until the window populates takes about a minute, regardless of whether sidebargui is installed or not.

Deed01, 2021-05-01:
tried to find a lead, 

1. i have remove all code i could from the plugin QML files and still get this LAG on menu and global GUI response (near all the code, only left structure declaration in qml files annd only SidebarContents.qml and sidebarstageMenu.qml still get few lines) (goal was to discard the most lines code i could from the problem) here it is:

[SidebarGUIPlugin_Cleaned.zip](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/6410307/SidebarGUIPlugin_Cleaned.zip)

2. without doing this, i notice that if i only removed Item at the end of SidebarContents.qml  the plugin work and doesn't disturb Cura GUI but i then miss the settings inside the right panel
[image]

3. Notice that the more extension/plugin are loaded and the more Sidebar will slow everything

Deed01, 2021-05-01:
Ho and there is warning in log concerning deprecated declaration of function inside connections 

It seams that it's corrected by changing this
    Connections
    {
        target: UM.Preferences
        onPreferenceChanged :
        {

        }

to this 

    Connections
    {
        target: UM.Preferences
        function onPreferenceChanged (preference)
        {

        }
    }

(don t' know QML, i google it => warning disappeared in log and  plugin still seems to work)
(must be done for all Connections call in the different QML files)

fieldOfView, 2021-05-01:
> It seams that it's corrected by changing this

I know that. Unfortunately making that change would mean the plugin will no longer work in previous versions.
A deprecation warning should not cause a slowdown. It is just a warning that this specific syntax will cease to work in some future version of Qt.

fieldOfView, 2021-05-01:
> Notice that the more extension/plugin are loaded and the more Sidebar will slow everything

I don't quite understand what you mean. Could you reword this?

Deed01, 2021-05-01:
I unloaded/deactivate all installed and all bundled plugin:
=> still get lag but a small one (manageable)
[image]

=> when a get all bundled plugin and installed one, it's clearly worst (could not found any specific plugin, seems juste to be the nember of them)

[image]

ps of course with all plugin activate and only sidebar unloaded everything work great

in the animation take a close look to the "over mouse" effect on menu (blue background menu hightlight)  => it's taking more time (as the menu showing up or panel opening..)

fieldOfView, 2021-05-02:
What installed plugins do you use? Perhaps if I install those I'll finally be able to reproduce this.

Deed01, 2021-05-03:
Installed plugin:
2021-05-03 09:17:09,567 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ArcWelderPlugin
2021-05-03 09:17:09,608 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CalibrationShapes
2021-05-03 09:17:09,645 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CustomSupports
2021-05-03 09:17:09,682 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GcodeFilenameFormat
2021-05-03 09:17:09,736 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MaterialSettingsPlugin
2021-05-03 09:17:09,783 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MeasureTool
2021-05-03 09:17:09,836 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MeshTools
2021-05-03 09:17:09,909 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevicePlugin.<module> [49]: Supplied version of Zeroconf module imported
2021-05-03 09:17:09,956 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin OctoPrintPlugin
2021-05-03 09:17:10,006 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin OrientationPlugin  => Auto-Orientation
2021-05-03 09:17:10,073 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ProfilAnalyser
2021-05-03 09:17:10,365 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SettingsGuide2
2021-05-03 09:17:10,403 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [425]: Plugin [SidebarGUIPlugin] has been disabled. Skip loading it.
2021-05-03 09:17:10,406 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [425]: Plugin [SidebarGUIPlugin] has been disabled. Skip loading it.
2021-05-03 09:17:10,447 - INFO - [MainThread] StartOptimiser.StartOptimiser.__init__ [40]: 1 containers are blacklisted by StartOptimiser
2021-05-03 09:17:10,452 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin StartOptimiser
2021-05-03 09:17:10,486 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin TabAntiWarping
2021-05-03 09:17:10,517 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin VersionInTitlebarPlugin
2021-05-03 09:17:10,549 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ZOffsetPlugin

fieldOfView, 2021-05-03:
Out of the plugins you listed, and the standard plugins I would think only the following could have any influence on the slowdown:

* `OctoPrint Connection` and `Ultimaker 3 Printing`, due responding to network traffic through bonjour
* `Settings Guide`, due to how it adds a lot of text and graphics to the sidebar tooltips
* `USB Printing`, due to how it continually monitors all serial ports

If you are still willing to do more testing, you could try disabling only those plugins.

In the mean time, I am also cleaning up the binding loops and deprecation warnings.

fieldOfView, 2021-05-04:
I have made a number of optimizations, but since I still can't reproduce the issue I don't know if I'm going in the right direction. 

Could you test this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/SidebarGUIPlugin/dheQkyBPJ3ahhUjjRCPPr1T6R-1kAst4T2J3N4EFqApV/SidebarGUIPlugin-v7.0.0-2021-05-04T13%3A37%3A43Z.curapackage?Expires=1935495473&GoogleAccessId=cura-packages-api-production%40stardust-193112.iam.gserviceaccount.com&Signature=QPdNwHeXcu270RYy5sCspa%2BmXZ%2Be%2FnhuKsjYL1BHIRCuPrMUNGcxi54ocS6vmKx6HqByVM9hCGNLgr%2F%2FDi0Sk88QuojbiNAjuI19Omzj8D4596SO6rFlZIsHWEoRk0AUzmWB8%2FDT2px5JpHmR3I%2BHr238OgCUShX8gnevx%2FXpMb1aLydLm2RQR%2FiqRLafUR6CrE04y9X%2BtJgoMHZiGBikAyGe50VLuRtt02RPwaet99NI0OecKEGQ5ub7ZcjsQgsLJocg2G7mmF0Pt3kaAjtOuzow59YAomzh%2FQYm6v8TcMOxNnKNsorheTkIu0eEll0EvvejH1pFheFh8zjxVhg7Q%3D%3D)? Download the .curapackage and drop it onto the Cura buildplate as if you were opening a 3d model. Then restart Cura as instructed.

Deed01, 2021-05-04:
Hi, sorry but still the same

Here is the Log
[cura.log](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/6421793/cura.log)

start by trying to move the 3DView, take 10s to get focus then it start moving and still menu lag
[image]

Deed01, 2021-05-04:
can t you make a special version with more log? or ca i activate something to get more detail in log?

With the plugin deactivate 4.1.2
[image]

fieldOfView, 2021-05-04:
> can t you make a special version with more log?

Honestly I would not know what to log.

Deed01, 2021-05-04:
May be putting timestamp to measure time taken here and there and try to locate part of the code taking more time on my computer than on your computer?

fieldOfView, 2021-05-11:
Could you test with [Cura 4.9.1](https://github.com/Ultimaker/Cura/releases/tag/4.9.1)?

brunoosti, 2021-05-14:
Had it in cura 4.9 and having it in 4.9.1.  Disabling sidebar "solves" the problem. 
https://github.com/Ultimaker/Cura/issues/9827#issuecomment-841250121
Arachne is fine and 4.8 was fine both with sidebar on.

fieldOfView, 2021-05-14:
I see that both of you are using an NVidia GPU. Perhaps it is GPU/driver related, because I am not using NVidia graphics and I am not seeing this issue. But I am grasping at straws here.

brunoosti, 2021-05-14:
I've just updated the drivers, but the sidebar still freezing cura. 466.27, GTX 970.

Deed01, 2021-05-14:
Currently using Driver 460.89 on GTX 1070

fieldOfView, 2021-05-15:
I tried with a GTX 1060 and driver 462.31, and it works for me, so it may not be the GPU afteral.

I'll see if I can get some more datapoints of other affected users. If there's two, there must be more.

fieldOfView, 2021-05-15:
> May be putting timestamp to measure time taken here and there and try to locate part of the code taking more time on my computer than on your computer?

The problem is that the plugin is mostly "markup", with limited scripting to move things around at the moment they are created. The best analogy I can come up with is HTML in a browser. There is no way to put a timestamp in the _HTML_, since it is the _browser_ that does the rendering. In case of this plugin, there are ways to "profile" the rendering of the markup (QML), but that would require me to create a special debug version of Cura (and you to set up and use another debugging application).

NoelStephaneon, 2021-05-18:
Same issue with v4.9 and v4.9.1, Disabled plugins one at a time to find which one was causing slow performance. Unfortunately it was the Sidebar GUI. Shame because It's a pretty awesome plugin. But 4.9x doesn't like it

fieldOfView, 2021-05-18:
@NoelStephaneon could you share a bit more about your system? OS, GPU. If at all possible, please run `dxdiag` on your system and post a link to the log it creates, so I can see if there is a common factor between affected users.

NoelStephaneon, 2021-05-18:
Ryzen 3700x, 64GB, RTX1060
DXDiag no issues   
[https://1drv.ms/t/s!AgmM8ZVrmDVLsGS_3ptBC2pU2AxT?e=fTXYEI](url)

fieldOfView, 2021-05-18:
@Deed01 and @brunoosti, if you could also share a dxdiag log, that could be helpful.

I have asked for feedback from more users [on reddit](https://www.reddit.com/r/3Dprinting/comments/ncttik/cura_sidebar_gui_plugin_needs_your_help/) and on the [Ultimaker forum](https://community.ultimaker.com/topic/36962-cura-sidebar-gui-plugin-needs-your-help/), but sofar I have not found a common ground (and none of the responses were from affected users). So I am no closer to reproducing or fixing this unfortunately.

brunoosti, 2021-05-18:
here
[DxDiag.txt](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/6503857/DxDiag.txt)

Deed01, 2021-05-18:
Hello, here is mine :

[DxDiag.txt](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/6504619/DxDiag.txt)

fieldOfView, 2021-05-19:
For a little while there I though that the `Citrix Indirect Display Adapter` that is on both @brunoosti's and @NoelStephaneon's log could be the silver bullit, but it is not in @Deed01's log. Hmm, the mystery continues.

brunoosti, 2021-05-19:
Tried disabling it and killing all the processes.... still the same thing in cura

Hyperspeed1313, 2021-06-19:
I'm [having a huge performance impact](https://github.com/microsoft/PowerToys/issues/11423#issuecomment-864346149) due to this plugin when used in combination with [Microsoft PowerToys](https://docs.microsoft.com/en-us/windows/powertoys/)' FancyZones (on the order of **minutes** to process an action). Don't know if it's the same root cause as the others' issues but it should be ***very*** easy to reproduce.

Steps to Reproduce:
1. Make sure Microsoft PowerToys is running (will have a system tray icon), with the FancyZones plugin enabled.
1. Launch Cura with Sidebar GUI installed & enabled
1. Click the "Open File" button or any of the drop-down menus in the menu bar.
1. Cura stops responding and has an infinite loading wheel for a mouse cursor.

Workarounds to Unbork:
* Disable FancyZones in PowerToys. As soon as it's re-enabled Cura will go back to being borked, but it's a quick way to get Cura responding to close it gracefully or disable the Sidebar GUI plugin.
* Kill Cura via Task Manager and restart it (but you can't really do anything because it'll start lagging before you get that far).

fieldOfView, 2021-06-21:
Thanks for trying.

fieldOfView, 2021-06-23:
There's briefly been a comment by @Hyperspeed1313, which the user has since removed, which mentions an issue of Cura with the sidebargui plugin enabled in combination with the Fancy Zones part of PowerToys. With Fancy Zones turned on, I can at least partially reproduce the slowness. Do any of you use PowerToys (or similar tweaks?).

brunoosti, 2021-06-23:
Clocked the link in the update email and couldn't find the comment and decided to leave it. 
So, yes. I disabled power toys and cura seemed to work better. Maybe not as responsive as 4.8, but surely a lot better than before. Usable again.

NoelStephaneon, 2021-06-23:
The Fancy Zones fix worked for me. Thank you for your efforts in identifying the culprit. I really missed this whilst I couldn't use it.

fieldOfView, 2021-06-23:
Awesome! @Hyperspeed1313 is the real hero here (also see https://github.com/microsoft/PowerToys/issues/11423) . Now that I have a way to reproduce this, *perhaps* I will be able to find a way to prevent the issue. So I will keep this open for a little while longer.

Hyperspeed1313, 2021-06-24:
So, deleted my previous comment because after I wrote it, I tried to reproduce the scenario on my second computer and couldn't get the slowdown behavior. Thought I'd hit an unreproducable anomaly because of that. Glad to have been proven wrong and that someone saw my comment before I deleted it.

fieldOfView, 2021-06-24:
> Glad to have been proven wrong and that someone saw my comment before I deleted it.

Email notifications!

Anyway, thanks for making the connection with FancyZones.

mikekscholz, 2021-08-04:
This appears to be fixed with todays release of powertoys, I've got fancy zones and sidebar enabled and its no longer locking up for me.

fieldOfView, 2021-08-24:
Since this is fixed in current releases of powertoys, I'm closing this issue.