fieldOfView/Cura-SidebarGUIPlugin #25: v4.1.2 in Cura v4.10 causes UI to freeze/hang Win10

quadcom, 2021-09-04:
I drilled down to this causing Cura to become unresponsive and pretty much unusable.

Fairly new to printing and Cura so if there are any logs or anything I need to provide, let me know where I can find those.

fieldOfView, 2021-09-04:
Do you have Powertoys installed? In that case, please look for an updated version of Powertoys.

quadcom, 2021-09-04:
As a matter of fact I do have powertoys. 

Currently running the latest version, it was just updated the other day.

fieldOfView, 2021-09-05:
See this issue: https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/24
Try disabling FancyZones

fieldOfView, 2021-09-06:
Did you get round to trying to disable FancyZones? Did that fix the issue?

quadcom, 2021-09-06:
Sorry, no I haven't. I'm pretty new to printing so I don't do a whole lot
and I use fancy zones all the time.

In the future, If you've figured out a fix where they can live together,
I'll definitely test it out.

fieldOfView, 2021-09-06:
I can't help you if you won't even temporarily turn the powertoy off to see if that indeed is the root of the problem. So I'll close this issue.

closed 2021-09-06 by fieldOfView