fieldOfView/Cura-SidebarGUIPlugin #26: profile selector dropdown overlaps view selection in older Cura's

rburema, 2021-09-07:
Plugin version: 4.2.0 (as submitted)
Cura versions: (at least) 4.0.0, 4.1.0

See screenshots:
[image]
[image]

fieldOfView, 2021-09-09:
Thanks for the report (and the diligence in testing)!

fieldOfView, 2021-09-09:
Turns out this has been an issue for a while. I'll fix it along with another minor issue with Cura 4.0-4.3 in plugin version 4.2.1.

hoonu, 2022-01-27:
> Turns out this has been an issue for a while. I'll fix it along with another minor issue with Cura 4.0-4.3 in plugin version 4.2.1.

Any updates on this? I'm using this plugin with the Creality Slicer which is based on 4.8 and it overlays the marketplace button.
[image]

fieldOfView, 2022-01-28:
Yours looks like a different issue. I am going to need a less-cropped screenshot.

I don't officially support Creality Slicer. It is more than enough work to support many different versions of Ultimaker Cura. Since Creality does not disclose what they changed in Cura, I can not test my plugin(s) with their changes.

fieldOfView, 2023-03-02:
I can't reproduce this anymore, so I am retiring this issue.

closed 2023-03-02 by fieldOfView