fieldOfView/Cura-SidebarGUIPlugin #27: Version 4.11 incompatibility 4.11

InnoProd, 2021-09-08:
Just upgraded to Cura 4.11, and sidebar docking appears to be non-functional
[image]
. No hurry, just thought I'd mention it.

Mitsuma, 2021-09-08:
If you want a quick workaround until fieldOfView brings out a proper update just go into:
"AppData\Roaming\cura\4.11\plugins\SidebarGUIPlugin\SidebarGUIPlugin"

Edit the plugin.json and change the "maximum_cura_version": "4.10" to 4.11

No warranty that 4.11 didnt change anything else that could break things, just in case somebody was as impatient as me.

InnoProd, 2021-09-08:
Works perfectly, and such a simple fix too. Thanks for the help!

fieldOfView, 2021-09-09:
Actually quite a bit changed in Cura 4.11, so I don't recommend this patch. I am close to releasing a version of the plugin for Cura 4.11, but before that is ready a development version that works in Cura 4.11 here: https://community.ultimaker.com/topic/26014-sidebar-gui-for-cura-4/page/5/?tab=comments#comment-294560

fieldOfView, 2021-09-09:
Version 4.2.0 of the plugin is now available in the Marketplace, and is compatible with Cura 4.11.

closed 2021-09-09 by fieldOfView