fieldOfView/Cura-SidebarGUIPlugin #29: Causes issues with 4.12.1

Machine73, 2021-12-09:
Been having issues with cura locking up whenever you add, click, or do pretty much anything. About a 10sec lock-up.
ie click open file...lockup. click quality....lockup. click material...lockup...etc.
I have tried numerous posted fixes on the net regarding cura lock-ups and crashes, but none fixed the issue. 
rolled back to 4.10...same issue.  removed all previous versions...same issue.
started disabling what few plug-ins I use and narrowed it down to sidebarGUI.
disabled and uninstalled it, problem went away. Cura works as normal.
re-installed it...problem comes back. I can have it installed and disabled with no issues, but once it's enabled again, lock-ups come back.
Cura has been uninstalled numerous times just to be sure.
I also installed the latest cura on my laptop, worked fine. added plug-ins 1 at a time leaving SBGUI till last. No issues at all until I install SBGUI then the lock-ups occur on laptop as well.

fieldOfView, 2021-12-10:
Do you have Microsoft Powertoys installed? If so, can you check if there is a newer version of it available?

fieldOfView, 2021-12-10:
Specifically, the FancyZones part of PowerToys seems to have caused this slowdown, at least until the august 2021 release. I;ve had reports that it was fixed then with a new release of PowerToys. See #24.

Machine73, 2021-12-10:
I sure don't. I try to keep my windows install to the bare minimums due to issues I've had in the past.
I did go through the your linked post and as "deed01" was showing, those are the exact same issues I am having. almost identical.
I scrubbed my logs and watched them populate as I was running cura. Didn't see anything unusual.

fieldOfView, 2021-12-10:
Well, then unfortunately I am as stumped as I was with the original issue. I have no idea what could be going on on your computer if it is not the same conflict with FancyZones.

Machine73, 2021-12-10:
I just tried your posted "theme fix" and didn't work, but on a hunch I changed my Cura theme from default to dark. Lag is still there but not as bad. 
The 1 thing I have in common with others is I have AMD ryzen and Nvidia gtx1070.
I will keep digging and I find anything I will post.

fieldOfView, 2021-12-10:
That theme fix has been part of Cura since many versions. It did not fix this issue then, it did not change anything now.

Unless you can reproduce a measurable difference between light and dark themes, I am afraid the difference is down to wishful thinking.

> The 1 thing I have in common with others is I have AMD ryzen and Nvidia gtx1070.

But the thing you don't have in common is that as far as I know those others were using FancyZones, and updating that fixed the issue. A Ryzen/GTX1070 is just a fairly common configuration.

G4Cab, 2021-12-25:
Hi, same problem here, Cura 4.12.1, Win11, Sidebar GUI 4.2.1
Opening, Multiplying or else operation locks Cura for several seconds.
after uninstalling all plugins it worked well again. Afterwards, i installed plugin after plugin again to find the culprit.
Sadly, the sidebar gui is the culprit.

fieldOfView, 2021-12-28:
The only thing I have for you is that if you use FancyZones, check for updates of PowerToys. Other than that I don't know what is causing this on your systems.

BerlinerFilialen, 2022-01-27:
Hello,

i got the same problem on cura, no matter the version. from 4.xx to 4.13
i thought the problem was caused by something else but on the point, i installed the 4.13 beta, it removed the plugin and after that, the Problem was gone. 
ich deinstalled it on every machine and my problem was gone. 
i reinstalled it and the problem was back. 
i just have a fresh installed win 10 and win 11 and i can reproduce the problem instant. 
right know i have to wait, feels like minutes to get the plugin removed from my last test.
its an intel i5 8250u with an nvida 130mx. it runs better if i use the nvidia for the graphik interface insteat of the intel. 
but i have the same issue on a ryzen 5 1600AF and also on a ryzen 5 5600x.

aaros-pl, 2022-02-07:
I have the same issue since update to 4.12.

fieldOfView, 2022-02-16:
Since I still cannot reproduce this without the old version of FancyZones, I need additional information to try to find out what is going on and stand half a chance to fix it.

If you can reproduce this issue with the SidebarGUI plugin enabled, please do the following:

* From the Marketplace, install the `Extensive Support Logging` plugin (by Ultimaker).
* Enable the SidebarGUI plugin (and restart Cura).
* In Cura, click on Extensions -> Remote Support -> Start CPU profiler from the application menu.
* Perform some task that is slow.
* Click on Extensions -> Remote Support -> Stop CPU profiler.
* Click on Extensions -> Remote Support -> Save remote support archive.
* Disable the SidebarGUI plugin (and restart Cura).
* In Cura, click on Extensions -> Remote Support -> Start CPU profiler from the application menu.
* Perform the same task as before.
* Click on Extensions -> Remote Support -> Stop CPU profiler.
* Click on Extensions -> Remote Support -> Save remote support archive.
* Post both resulting archives here.

This will hopefully get me more information about what it is in my plugin that makes things slower.

akamagnum357, 2022-03-19:
yay, i removed the sidebar plug in, and everything started working again. yay. ty.

fieldOfView, 2022-03-19:
I don't mean to sound saly about this, but by just removing the plugin and not providing me with the data I requested, you are all ensuring this will likely never get fixed.

G4Cab, 2022-03-19:
I've created your wanted logs.
Maybe i found a clue, if i use my printerprofile with multiple tools it's damn slow with sidebar activated, on my printerprofile with only one tool it's not this remarkable.

https://yoruichi.golf-4-cab.de/nextcloud/index.php/s/GbqB4pq55pLJz7D

fieldOfView, 2022-03-19:
Thanks for the logs. I will investigate.

What do you mean by "multiple tools"? Multiple extruders?

fieldOfView, 2022-03-19:
@G4Cab, in all of the logs, there are a lot of errors caused by the Repetier output device plugin. Have you tested without that plugin?

G4Cab, 2022-03-19:
> 

Yep, i had all plugins removed (fresh cura install) and started to add plugin by plugin.
sidebargui was one of the first i installed, cause i love the sidebar instead of this floating settings window
repetier followed later

fieldOfView, 2022-06-20:
Can someone test if this is still an issue with Cura 5.0?

BerlinerFilialen, 2022-06-20:
Hi, 

i tried it yesterday and i got the same issue. i need 7-8 mins to deinstall the programm. i think, maybe its an programming enviroment, whats missing.

if i install it, it needs hours to register any input. 

i Installed it on an HP Pavillion Laptop with 16GB ram and an NVME m.2

BerlinerFilialen, 2022-06-20:
@fieldOfView 
i made the two archives. one with, one without. hope it helps.

i miss this plugin soooo much.

[With-sidebar-remote-support_22-06-20-23_19_03.zip](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/8943730/With-sidebar-remote-support_22-06-20-23_19_03.zip)
[Without-sidebar-remote-support_22-06-20-23_19_03.zip](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/8943731/Without-sidebar-remote-support_22-06-20-23_19_03.zip)

fieldOfView, 2022-07-13:
> I do use FancyZones but that's a nonstarter for me to remove it. Id rather do without the plugin vs without FancyZones, but that's just me.

Even if that would be a non-starter as a workaround, I would still very much appreciate if you could let me know if (temporarily) disabling FancyZones makes any difference.

MagFlux, 2022-10-11:
Understood, I’ll give it a go when I get home from work and report back.
Thanks for the prompt response!

TWIISTED-STUDIOS, 2023-01-03:
Ok, so I recently noticed issues in cura where it would load and let me move around but if I clicked any menu it would crash lag and just become unresponsive same with loading in a model, I had no idea what it was causing it, but came here read that users of the side bar plugin, and powertoys was an issue, so I closed powertoys and long behold the problem was gone. Boot up power toys again and the issue came back, but I can't go back to the default GUI its just so clunky so I have just to disable powertoys when opening cura from now on I guess.

marcribe, 2023-01-04:
Same here with PowerToys...

fieldOfView, 2023-02-02:
I can now reproduce the issue with Powertoys. Even with the latest version, which according to its release notes includes a refactored/cleaned up Fancy Zones, there is a clear drop in responsiveness.

Observations:
* Turning off Fancy Zones fixes the problem immediately; no restart or Cura or PowerToys is necessary
* The effect is greatly affected by the number of "visible" settings; hiding all settings almost mitigates the responsiveness issue almost entirely (but not 100%).
* Closing or undocking the sidebar does not make much of a difference

My conclusion is that there seems to be a lot of unnecessary (re-)painting of parts of the UI going on, even for settings that are not currently visible on screen. I do not know if this is somehow triggered by FancyZones, or if FancyZones does something that merely makes (re-)painting the elements take slightly longer (and many times "slightly longer" builds up to "a lot longer"). It is strange that closing the sidebar - which should stop setting control from being drawn - does not seem to alleviate the problem much; apparently even when not shown, the controls are somehow getting drawn.

The puzzling part for me is that my plugin does not change how the settings are drawn, just where they are drawn. It takes them from one "parent" element and puts them in another "parent" element. And ofcourse why this would be affected by FancyZones at all.

fieldOfView, 2023-03-13:
> I can now reproduce the issue with Powertoys.

And now I can't anymore. I'm testing on my laptop, which normally uses an Intel Iris Xe iGPU. Tomorrow I can test if using the same laptop with an NVidia eGPU reproduces the issue.

fieldOfView, 2023-03-13:
I just found a very similar report with the SettingsGuide plugin:
https://github.com/microsoft/PowerToys/issues/24758 (and the report for the Settingsguide: https://github.com/Ghostkeeper/SettingsGuide/issues/111)

fieldOfView, 2023-03-14:
After more testing, I am again puzzled. 

I can reproduce the issue with Powertoys FancyZones enabled _and_ the driver for my Wacom tablet loaded. If I start Cura before I plug in the tablet, I have no issues, even when running FancyZones. Plug in the tablet before starting Cura, and the lag reproduces. 

😕

fieldOfView, 2023-04-04:
There is a "nightly" build of Cura available that seems to fix this issue. If any of you feel adventurous, The build includes a newer version of the QT gui framework. I could use some feedback with that build. 

* Download the Cura build from here:
https://github.com/Ultimaker/Cura/discussions/15098
* Open the Cura configuration folder for version 5.4 (Help -> Show configuration folder)
* Go to `plugins\SidebarGUIPlugin\SidebarGUIPlugin`, and open `plugin.json` in an editor
* Change the `maximum_cura_version` from `5.3` to `5.4`
* Restart Cura

TWIISTED-STUDIOS, 2023-04-10:
> There is a "nightly" build of Cura available that seems to fix this issue. If any of you feel adventurous, The build includes a newer version of the QT gui framework. I could use some feedback with that build.
> 
> * Download the Cura build from here:
>   [A Nightly Build with an updated PyQT Library Ultimaker/Cura#15098](https://github.com/Ultimaker/Cura/discussions/15098)
> * Open the Cura configuration folder for version 5.4 (Help -> Show configuration folder)
> * Go to `plugins\SidebarGUIPlugin\SidebarGUIPlugin`, and open `plugin.json` in an editor
> * Change the `maximum_cura_version` from `5.3` to `5.4`
> * Restart Cura

ill give it a try for you this week

TWIISTED-STUDIOS, 2023-05-28:
> There is a "nightly" build of Cura available that seems to fix this issue. If any of you feel adventurous, The build includes a newer version of the QT gui framework. I could use some feedback with [...]

This fella is working perfectly its been tested now for a good few weeks and has had no issue yet. so whatever is fixed it now works with that build mentioned above.

fieldOfView, 2023-07-04:
It looks like this is fixed with Cura 5.4

closed 2023-07-04 by fieldOfView

MagFlux, 2023-07-06:
Can confirm, with both 5.1 and 5.4 installed. 5.1 has issue, 5.4 issue is resolved.