fieldOfView/Cura-SidebarGUIPlugin #30: Does not work with Cura 4.13.0 Beta

mbourque, 2021-12-27:
[image]

DeAlexPesh, 2021-12-27:
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/blob/release/plugin.json

...\AppData\Roaming\cura\4.13\plugins\SidebarGUIPlugin\SidebarGUIPlugin\plugin.json

```
- "maximum_cura_version": "4.12",
+ "maximum_cura_version": "4.13",
```

mbourque, 2021-12-27:
Fixed it. Thank you.

fieldOfView, 2021-12-28:
I have published a new version of the plugin to the marketplace, but it is awaiting approval.

Fixed in https://github.com/fieldOfView/Cura-SidebarGUIPlugin/commit/cf77054970cdd40e2f7e7553d3d79fd7b2eeec96

fieldOfView, 2021-12-28:
I have added a note to the readme:
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/blob/master/README.md#cura-version-compatibility

mbourque, 2021-12-28:
> Fixed in cf77054

Thank you

fieldOfView, 2021-12-29:
Update is now available in the Marketplace

closed 2021-12-29 by fieldOfView