fieldOfView/Cura-SidebarGUIPlugin #31: The current version of the plugin does not work with a new Cura release

fieldOfView, 2021-12-28:
When a new release of Cura is available, the Sidebar GUI plugin always needs to be updated.

In order for the Sidebar GUI plugin to work, it does quite a bit of "nasty" patching of the original interface. These patches may break the interface when the original interface is changed. So, as a security measure, the plugin checks to see if it has been "approved" for the version of Cura you are using it with.

I normally work hard to have an updated version of the plugin ready during the beta period, so when a new *stable* release of Cura is available, the plugin update is available in the Marketplace. Because I may not always have time immediately when a Cura beta is released, and because updates to plugins need to be checked and approved by Ultimaker developers, there is always a bit of a delay.

Thisismydigitalself, 2022-04-24:
SidebarGUIPlugin_v4.3.0-DEV_Cura5.0 works for me. thank you Aldo.

fieldOfView, 2022-07-09:
It is that time of the month again with a new beta of Cura out. I have created a new prerelease that is compatible with Cura 5.1. I don't have a lot of time to test it myself right now, so it would be nice if some of you could install and test this prerelease:

https://github.com/fieldOfView/Cura-SidebarGUIPlugin/releases/tag/v4.4.0-DEV

Download the [SidebarGUIPlugin_v4.4.0-DEV_Cura5.0.curapackage](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/releases/download/v4.4.0-DEV/SidebarGUIPlugin_v4.4.0-DEV_Cura5.0.curapackage), and drop the resulting file onto the buildplate in Cura as it you were opening a 3d model.

(note to self: fix packager so it is clearer that the release is for 5.0 and up)

Thisismydigitalself, 2022-07-10:
Just installed 4.4.0 on 5.1 and we're back in business.
thank you Aldo.
your plugs are used by hundreds of thousands of users and this alone tells me that Cura team should take that into account each time they release a new version. 

Keep up the good work.
Cheers.

fieldOfView, 2022-07-10:
Thanks for the kind words, @Thisismydigitalself

pnt103, 2022-07-22:
I just installed 4.4.0-DEV, and that fixed things for me too.  Thank you!

yet-another-average-joe, 2022-10-27:
There's an issue with Cura 5.2.1 : Sidebar GUI does not load, and Cura asks for it to be uninstalled...
I don't know for Cura 5.2.0 (5.2.1 came out quickly so I had no time for testing)

It seems Cura Marketplace does install an older version of Sidebar GUI (4.4.0).

Found a quick and dirty workaround : install Sidebar GUI 4.4.0, and do *not* uninstall it ; close Cura, download 4.4.1-DEV and overwrite the files in the plugin directory (in the configuration folder). Restart Cura.

It seems 4.4.1-DEV does work as expected with Cura 5.2.1 ; 4.4.1 DEV displays as 4.4.0 in the Marketplace. No idea how to install the plugin properly, but it works after doing that...

erickwill, 2023-12-02:
Looking forward for a new update in order to fix it to work along with the new cura version 5.6.0

yet-another-average-joe, 2023-12-02:
Thanks. Wasn't notified for 5.6.
5.5 was such a disaster that I immediately reverted to 5.4.0.
5.5, the "unable to slice" slicer ; but slicing when changing a bit any parameter. While Cura is the best slicer, by far. Slic3r descendants are toys, Cura being the only real tool. And Sidebar the add on that makes it usable ; should be mainstream.

Sorry for the rant.

Will no install 5.6 until Sidebar supports it. I stay tuned.

fieldOfView, 2023-12-04:
If you actually read the opening post, it has a link to this page:
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/releases

That page has a prerelease of the plugin that is compatible with Cura 5.6.

GregValiant, 2025-03-19:
Hello Aldo.
I adjusted the version number in SidebarGUI (and TabbedSettings) so they would load in 5.10beta.  (That is my normal routine and hasn't been a problem.)

This could just be me being impatient, but here is what I've noticed.

In 5.10beta-1:

- TabbedSettings is throwing QML errors and isn't loading.

In regards to SideBarGUI...
- The "printer box" is not populated with the installed printers.  I have to use "Settings/Printers/Manage printers" to change machines.  The selected printer correctly appears in the text area of the dropdown (but the dropdown list is empty).

- The "profile box" seems to have issues.  The original box remains but the selected profile does not appear in the text area of the dropdown (but the list is populated).
- The original "profile box" remains and works.

[image]

fieldOfView, 2025-03-19:
That must be why I don't have a beta yet... More work than just upping the version number is needed this time, and this is exactly why I have implemented the version check in the first place.

fieldOfView, 2025-03-25:
@GregValiant, please test the new beta release with Cura 5.10:
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/releases/tag/v4.6.8-DEV

GregValiant, 2025-03-25:
Thanks you.
I just installed 5.10 stable and SidebarGUI looks good.
Now you only have about 20 more to go.

joseportillodev, 2025-03-27:
I installed manually the v4.6.8-DEV on my recently updated Cura 5.10 and worked perfectly, as before. Today Cura prompted me to update to v4.6.8 and it also works as expected.

Thank you so much! This extension should come standard with Cura or be the default layout.

ManuMod, 2026-08-31:
I confirm if you modify the plugin.json, It work.
You can find it in this folder: ...\AppData\Roaming\cura\5.13\plugins\SidebarGUIPlugin\SidebarGUIPlugin
You need to modify the line 6, like this : "maximum_cura_version": "5.13",

Careless, 2026-09-25:
> I confirm if you modify the plugin.json, It work. You can find it in this folder: ...\AppData\Roaming\cura\5.13\plugins\SidebarGUIPlugin\SidebarGUIPlugin You need to modify the line 6, like this : "maximum_cura_version": "5.13",

thank you for sharing.

for macOS users, it is located here:

`'/Users/<YOUR_MAC_USERNAME>/Library/Application Support/cura/5.13/plugins/SidebarGUIPlugin/SidebarGUIPlugin/plugin.json'
`

I just changed it to 5.15 so it doesn't bother me for a while. 🫠

PartySausage, 2026-09-26:
> thank you for sharing.
>
> for macOS users, it is located here:
>
> `'/Users/<YOUR_MAC_USERNAME>/Library/Application Support/cura/5.13/plugins/SidebarGUIPlugin/SidebarGUIPlugin/plugin.json' `
>
> I just [...]

FWIW The plugin won't work with 5.14 which is due to be released soon. So if you install that you will get some issues with you having the plugin set to version 5.15

Careless, 2026-09-26:
> FWIW The plugin won't work with 5.14 which is due to be released soon. So if you install that you will get some issues with you having the plugin set to version 5.15

fiiiiiiiiiiiiiiine... i'll change it 5.13 😔

PartySausage, 2026-09-26:
> fiiiiiiiiiiiiiiine... i'll change it 5.13 😔

😊 I have a fork that works with the Cura 5.14-Alpha build, but it's heavily interlaced with the TabbedSettings plugin & there's some biiiig changes coming regarding that feature in the official Cura 5.14 release that I won't be able to test until it's released so there's a good chance even my fork won't work with the official release

dxgldotorg, 2026-09-27:
Doesn't 5.14 upgrade the Qt version?

PartySausage, 2026-09-27:
> Doesn't 5.14 upgrade the Qt version?

Yes I think it's going from Qt 6.6 to 6.10

Careless, 2026-11-08:
@PartySausage anything promising to report with cura 5.14's release?

PartySausage, 2026-11-08:
I have a version that works on 5.14, which has been tuned to my requirements so it may look a little different to the original. It also requires a modification to one of the core Cura QML files to resolve an issue with the 3MF Import dialog when opening a project file from a windows file association.

First and foremost this plugin belongs to @fieldOfView & it wouldn't be ethical for me to distribute my version widely without his permission. If @fieldOfView was to reply here giving me permission then I will attach my version with the instructions on the Cura QML file to replace. If permission isn't granted then I'm afraid I cannot provide my version

dxgldotorg, 2026-11-08:
@PartySausage You can legally distribute a modified version of the plugin ad long as you comply with the license - https://github.com/fieldOfView/Cura-SidebarGUIPlugin/blob/main/LICENSE

PartySausage, 2026-11-09:
OK I've checked the GitHub Licencing agreement & don't think I'm violating anything by posting so here's my personal version of the SideBarGUIPlugin, which has been adapted to integrate with both Cura 5.13 & 5.14. 

I've fixed a lot of legacy issues with the Plugin/Cura UI Interaction & made it compatible with the new Cura Tabbed Settings Layout. I've also made some tweaks to the UI to suit my own personal requirements, which may not be compatible with every users setup
All of my changes are annotated in the respective files so if you don't like any of the changes then have a play to find a layout that suits your needs
 
Also in the attached zip file is a tweaked version of the core Cura.qml. This fixes a plugin conflict when Cura is opened from 3MF Project, Windows Explorer file association launch, which was introduced in Cura 5.13 and results in the Project Import dialog either displaying then disappearing or the Import dialog displaying with a white title bar & buttons rendered offset from their click area when the UI loads

If you experience those issues then you can drop the file into the **\UltiMaker Cura 5.14.0\share\cura\resources\qml folder**. 
I've also included the original Cura.qml file from the 5.14.0 installation in case you want to revert back. (Do so at your own risk, It works in my own setup but can't guarantee it won't break your Cura installation)

I have no wish to take over as the maintainer of this plugin nor am I planning on addressing any bugs that may be generated in an individuals setup. This is my own personal endeavour that has been tested across Cura 5.13, 5.14 Alpha, Beta & Stable Releases. Each of, which has introduced more conflicts that have needed to be resolved with each release & has sometimes been a right pain in the ar5e to sort out

Good Luck....

[SidebarGUIPlugin.zip](https://github.com/user-attachments/files/34595968/SidebarGUIPlugin.zip)

GregValiant, 2026-11-09:
Here is a toast to pain in the arses.
I've been one all my life.