fieldOfView/Cura-SidebarGUIPlugin #32: Material selection is outside the dialog border and not accessible

GregValiant, 2022-05-01:
Hello A. Hoeben.  Nice place ya have here.  I was worried you might be getting bored so I brought you something to think about (although it may be that I have an early fix on the plugin and this has been taken care of).

I had my printer set up as a dual extruder and went to change materials on Extruder 2.  The drop down was OK.  I selected the manufacturer and the secondary selections appeared but the third set was off to the right and I couldn't get to them to actually change material.  Undocking the sidebar didn't help.  Here are two images with the sidebar docked and undocked.
The third image is from 4.13 where the primary choices went to the left and selecting the material worked.
(I used the Settings menu and successfully changed material.)

[image]
[image]
[image]

fieldOfView, 2022-05-03:
This is a disadvantage of replacing native controls with "homebrewn stuff", the way Cura 5 does it. A similar problem happens without the sidebar gui plugin. I'm going to keep an eye out if this upstream patch fixes it:
https://github.com/Ultimaker/Cura/pull/11977

GregValiant, 2022-05-03:
Thanks for looking at it.
I know the new release has been tedious for you and the Cura team but overall it looks like things are going well.  There has been a lot of interest.

5axes, 2022-05-04:
I just take advantage of this issue to ask if it would be possible to have also the color of the material as in the material parameters.
Example in version 4.13
[image]

In the Material settings

[image]

5axes, 2022-05-17:
Tested with the "Official" release this issue is still present

fieldOfView, 2022-05-27:
Could you test this prerelease?
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/releases/tag/v4.3.0-DEV

GregValiant, 2022-05-27:
That did it.  I'll close this.

closed 2022-05-27 by GregValiant