fieldOfView/Cura-SidebarGUIPlugin #36: discussion: sidebar scrollable area, scrollbar size and scrolling behavior

StuSerious, 2022-06-15:
Hey there Aldo, wanted to get your two cents on the scrollbar behavior / width.

Currently the 'scrollable area' is limited to the rightmost side of the scrollbar itself.
This means that when the pointer is immediately to the right of the scrollbar, it will stop scrolling the sidebar and start zooming the viewframe instead.
Let me give you a visual since I'm QML illiterate and I suck at explaining:

[image]

Is this expected behavior?
I find it a bit uncomfortable since I'm so used to slamming the cursor out of view to the right of the screen to scroll (like you do in web pages or MS Word for example).
I think the 'scrollable area' should be extented to the rightmost side of the window.

This is just nitpicking and something I can live without, but wanted to put it out there to see if it had some purpose that i'm missing or if it's a super ez fix.
Thanks for creating and maintaining one of the best suite of plugins there is out there for **_any_** piece of software!

Cheers!
_~S²_

fieldOfView, 2022-06-20:
> Is this expected behavior?

No, I consder this a bug.

fieldOfView, 2023-03-18:
Sorry it took me a while, but I think I have finally found a way to fix this for the next version of the plugin.
See [172d783](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/commit/172d78373defeeffc9befcdcb6d3923e67a5ad96)

commit 172d7837 (2023-03-18): Extend scrollbar to the far right of the window

Fixes #36