fieldOfView/Cura-SidebarGUIPlugin #37: Sidebar GUI 4.3.0 makes Cure 4.13.1 and 5.0 lag, stop responding, and is unusable.

gleep52, 2022-07-04:
I haven't used cura in a while - but it was working great - stayed on 4.13.1 for a long time since it was working top notch.  Took a few weeks off from dabbling in 3D - few days ago opened cura and it was just a mess to try to load files, scroll menus, etc.  Long "not responding" messages and weird stuff happening.  Disabled AV, reboot, reinstall, etc. no dice.

Finally disabled plugins one by one until I found Sidebar GUI 4.3.0 was the culprit.  I don't know if I had updated it prior to figuring it out that it was the "bad one" in my setup, but for sure now it is the only plugin I have disabled and everything is still top speed - but if I enable Sidebar GUI 4.3.0 - restart cura - the lag monster is back.  I installed 5.0 just to see if it was only an update that works with 5.0 and that has the same lag problem too.  Just checking in to see if this was common and if there is a work around.  I love my sidebar GUI :)

fieldOfView, 2022-07-04:
I'm closing this as a duplicate of #29. Thanks for the additional data that this is still an issue with Cura 5 and the latest version of the plugin.

closed 2022-07-04 by fieldOfView

fieldOfView, 2023-07-04:
It looks like this is fixed with Cura 5.4