fieldOfView/Cura-SidebarGUIPlugin #38: Nozzle and material choice boxes aren't visible.

GregValiant, 2022-10-03:
When clicking on the arrow to show the Nozzle size and Material selection boxes - the area remains blank.
5.1 on the left and 5.2 on the right.
[image]

fieldOfView, 2022-10-03:
Thanks, fixed in https://github.com/fieldOfView/Cura-SidebarGUIPlugin/commit/48bad5e98812757ba47b6bec877912ed265a3feb

This sort of bug happens when in a new version of Cura QML elements are rearranged. In this case an element was removed, so instead of the material/nozzle selectors the plugin moved those extruder tabs there, which are right above the material/nozzle selectors in the original UI. See https://github.com/Ultimaker/Cura/commit/b7fd75b2dd202f2516f9188e342eccc8a15ae6ab

I missed that while checking for compatibility because I had the configuration collapsed.

GregValiant, 2022-10-13:
After much pulling of hair and biting of fingernails I made the changes.  The fixes seem to be working fine.

closed 2022-11-02 by fieldOfView