fieldOfView/Cura-SidebarGUIPlugin #40: Cura 5.1/5.2

WimVDK, 2022-10-23:
Due to high energy prices my 3d printing was a bit on the sideburner (still is) but i'm updating things atm. Just updated to cura 5.2 and tried to reinstall the sidebar GUI. The official released one 4.4.0 still isn't compatible, however i noticed a 4.4.1 DEV version under releases that's dated about a week ago. However its in .curapackage, never used that before. Can you elaborate on how i can update to the dev version?

dcervenkov, 2022-10-24:
[Apparently](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/31#issuecomment-1179494228) you just drag & drop the .curapackage on the build plate like you would an STL file.

fieldOfView, 2022-10-24:
What @dcervenkov says.

The new version has been submitted to the Marketplace for a couple of weeks now, but it is still in the review queue at Ultimaker.

I'm closing this as a duplicate of https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/31

closed 2022-10-24 by fieldOfView