fieldOfView/Cura-SidebarGUIPlugin #41: Cura 5.3 Beta Missing Icone to access to teh reset and Save Function

5axes, 2023-02-24:
In Cura 5.3 the UI have change and Now Reseting or Saving the modification into a project is done via some Icone on the Side of the profil : 
[image]

And Not in the Drop Down Menu as it was before. 

[image]

So with the SideBarPlugin we don't have anymore acces to this function , We need to open the Profile Manager to Update Profil settings

fieldOfView, 2023-03-02:
I fixed this by replacing the "star" icon with the two new icons.  That star icon should never have been a star icon in the first place.

5axes, 2023-03-02:
Thanks a lot .

Works in 5.3 Beta 2
[image]

And 5.3 Alpha Xmas
[image]

fieldOfView, 2023-03-03:
I'll close this when a new release is available in the Marketplace

closed 2023-03-16 by fieldOfView