fieldOfView/Cura-SidebarGUIPlugin #42: Material selection from side bar not working.

3Dkat, 2023-03-14:
Cura 5.2.1 when trying to select material from the side bar, first two levels of the material menu are opening fine (manufacturer and material type), where color selection is no longer opening for neither custom nor default materials. Clicking on material type does not do anything.  In my case same story for single and dual extruder machines. Selection is available only from settings menu. Updated version of plug in works fine in version 4.13.0

Thank you

fieldOfView, 2023-03-16:
Here's someone else affected by the bug:
https://community.ultimaker.com/topic/43098-problem-selecting-materials-in-cura-522

They confirmed that using the previous version of the plugin fixes the problem:
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/releases/tag/v4.5.0
Download the .curapackage and drop it onto the Cura buildplate as if you were opening a model.

3Dkat, 2023-03-16:
> Here's someone else affected by the bug: https://community.ultimaker.com/topic/43098-problem-selecting-materials-in-cura-522

> 
> They confirmed that using the previous version of the plugin fixes the problem: https://github.com/fieldOfView/Cura-SidebarGUIPlugin/releases/tag/v4.5.0 Download the .curapackage and drop it onto the Cura buildplate as if you were opening a model.

Sorry was out for some time and missed a message. Well its the same issue as with Carbon in this thread https://community.ultimaker.com/topic/43098-problem-selecting-materials-in-cura-522
Same exact thing. Thank you for the solution, was thinking on rolling back, but decided to wait and here it is. If anything else is needed, I am here.

fieldOfView, 2023-03-16:
Could you try if this release works for you in Cura 5.2?
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/releases/tag/v4.6.1-DEV

3Dkat, 2023-03-16:
Works like a charm!  
Works with Cura 5.2. 3ed level menu appears, colors are showing.
Previously I said that its working fine in 4.13.0 but it was missing color dots next to the filament name. Still missing with this update. Menu works thou.

fieldOfView, 2023-03-16:
The color dots will only work with Cura 5.0 and newer.

I'll mark this issue "fixed", but will keep it open until a new version of the plugin is available in the marketplace.

3Dkat, 2023-03-16:
Please hold! Just now I was slicing a model and I am getting an issue with slicing have failed. Never happened before.
This happens when I change initial layer setting form 0.2 to 0.3

3Dkat, 2023-03-16:
Here is what happens. I open previous project. Sometimes right from the start it gives me "unable to slice" even thought I printed it yesterday, and that was before I updated. I change the color of the filament and it slices. Next color does not. Those are my custom materials. Also when I change initial layer height it will not or will slice (honestly can not follow the pattern of events). 
[image]
[image]
I also checked a possible issue with start up optimizer plug in. Uninstalled it, but issue still on hand.

fieldOfView, 2023-03-17:
You can't blame _everything_ on my plugin ;-)

Just looking at the reported issues at the main Cura issue queue, there are lots of people suffering from the "slicing failed" problem:
https://github.com/Ultimaker/Cura/issues

If you move the model away from the mid-point of the Cura build plate the model will probably slice.

closed 2023-04-19 by fieldOfView