fieldOfView/Cura-SidebarGUIPlugin #44: Cura 5.6 support

dxgldotorg, 2023-12-10:
Currently, the plugin fails to load in Cura 5.6.  Please update and upload to the Marketplace at your convenience.

fieldOfView, 2023-12-11:
Please read this:
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/31

A prerelease is available. A fixed version is awaiting approval from Ultimaker. My hands are tied.

closed 2023-12-11 by fieldOfView

erickwill, 2023-12-17:
> Please read this: #31
> 
> A prerelease is available. A fixed version is awaiting approval from Ultimaker. My hands are tied.

Is there any way to install it apart of ultimaker marketplace? I didn't find the proper version here in git..

fieldOfView, 2023-12-18:
Sigh... https://github.com/fieldOfView/Cura-SidebarGUIPlugin/releases/tag/v4.6.4-DEV

Download the .curapackage, drag and drop it onto the Cura viewport as if you were opening a 3d model.