fieldOfView/Cura-SidebarGUIPlugin #45: SidebarGuiPlugin_v4.6.4 will not load with Cura 4.13.1

DivingDuck, 2023-12-21:
Hi @fieldOfView,
FYI.
I had the need to run Cura 4.13.1 and just recognize after updating some plugins that version 4.6.4 do not load and is deactivated after restarting Cura.
I tested the different plugin versions from here and the latest working version was SidebarGUIPlugin_v4.6.2_Cura4.4-4.13.curapackage. All newer versions gave the same information as shown below.

SidebarGUIPlugin_v4.6.4_Cura4.4-4.13.curapackage:
[image]
[cura.log](https://github.com/fieldOfView/Cura-SidebarGUIPlugin/files/13746031/cura.log)

fieldOfView, 2023-12-22:
I will investigate why the plugin does not start in Cura 4.13.

In the mean time, you can download the previous version of the plugin here:
https://github.com/fieldOfView/Cura-SidebarGUIPlugin/releases/tag/v4.6.3

Download the .curapackage for your version of Cura and drag and drop the resulting file onto the buildplate in Cura, as if you were opening a 3d model. Then restart Cura as instructed.

DivingDuck, 2023-12-22:
Hi,
as mentioned, v4.6.3 have the same problem. I had reinstalled v4.6.2 as this version is working. So its not a real problem as I can take that version as long as the newer versions do not work. 
I wish you a nice holiday season.
Best regards,
DD

DivingDuck, 2023-12-22:
You are right 4.6.3 works. It seems I was confused with versions during testing.

DivingDuck, 2023-12-22:
I just test https://github.com/fieldOfView/Cura-SidebarGUIPlugin/commit/75936eb8181a341d2d76421f814965e2a3c74776. I replaced the file SidebarIncompatibleVersion.py in the actual 4.6.4 version manually. The sidebar is working with this change 👍

pierremilcent, 2023-12-23:
4.6.4 DEV does work for me, while 4.6.4 does not load.

fieldOfView, 2023-12-23:
@pierremilcent That is correct. The change I made that causes the issue in 4.6.4 was not yet in 4.6.4-DEV. So you can either use 4.6.3 or 4.6.4-DEV, they are functionally identical for Cura 4.x.

pierremilcent, 2023-12-23:
@fieldOfView Yes this is what i did : reinstall the 4.6.4 DEV.

fieldOfView, 2023-12-23:
I am marking this issue as "fixed" (because a fix is available in https://github.com/fieldOfView/Cura-SidebarGUIPlugin/commit/75936eb8181a341d2d76421f814965e2a3c74776), but I am leaving this issue open until a fixed version of the plugin is available in the Marketplace so other people can find this issue and see the workaround.

closed 2024-10-17 by fieldOfView